ericklein/air_quality #32: PM 2.5 sensor integration

ericklein, 2022-01-27:
Integrate PM 2.5 sensor data

disquisitioner, 2022-03-14:
Will be good to sync on the preferred approach here.  The most direct way would be to connect a PM2.5 sensor via I2C and read data from it alongside the CO2 sensor.  This should be easy electrically (and in software) but introduces a potentially significant power drain due to the airflow fan in the sensor plus the need to run the sensor for a short period to start getting reliable readings.   Also would need a larger case with more airflow, but that might allow a larger battery and mitigate some of the power concerns.  But all doable.

An alternative would be to design PM2.5 measurement as a companion device that is optimized for handling typical PM2.5 sensors and perhaps would be in a different location.  The sensor and reporting code could largely be the same as the directly-integrated case. 

Having a stand-alone PM2.5 unit at least for development and testing would all exploring power consumption and sampling cycle constraints, so will start there while we talk through the overall design approach.

disquisitioner, 2022-06-04:
As an update, I have a separate code base using an Sensirion [SEN54 particulate sensor](https://www.sparkfun.com/products/19325) with I2C intervace, running on an Adafruit Feather, uploading data both to Dweet.io and my InfluxDB backend.  As a starting point I'll create a public repo with that code so we can take a look and see how to align the implementation there with infrastructure we've already worked out here for air_quality.

disquisitioner, 2022-06-05:
My current PM2.5 AQI sensor project code is now on github at [https://github.com/disquisitioner/Feather-AQI-Monitor-SEN54](https://github.com/disquisitioner/Feather-AQI-Monitor-SEN54).   Issues, new feature ideas, and pull requests are welcome!

closed 2022-06-09 by ericklein