ericklein/air_quality #45: post_dweet() Ethernet support

ericklein, 2022-03-14:
post_dweet() has WiFi connection check but not for Ethernet. It might be better to say we need a generic handler to check Internet status?

disquisitioner, 2022-03-15:
Good point.  We have two physical device configurations via #define for WiFi and wired, and then one global state variable that says whether there's an active connection or not that gets used in an if test inside that compiled code.   I'll take a look at ways to make that more straightforward given there are few generalizable functions that could be called independent of WiFi vs. wired.

ericklein, 2022-05-01:
Closing this is dependent on headless execution, which is currently not working. I'll work on the headless issue as a branch from master.

disquisitioner, 2022-05-12:
I think this should work now given the introduction of a generalized networking interface that handles WiFi and Ethernet connectivity identically.   Will be good to test to make sure that's the case, though.

disquisitioner, 2022-06-04:
The recent generalization of the networking code (via aq_network.cpp) enabled post_dweet() and post_influx() operations with Ethernet as well as on WiFi.  This bug is therefore fixed, though it would be good to test it with working Ethernet hardware (hard wired, not WiFi).

closed 2023-02-27 by ericklein