ericklein/air_quality #47: DWEET support for when CO2 sensor is not available

ericklein, 2022-03-22:
When CO2 sensor is not available, AQ uses the convention of setting sensorData.internalCO2 to 10000. DWEET will attempt to push CO2 values even when there is no sensor reading it, so it will push 10000 upstream. Is that the intended behavior?

disquisitioner, 2022-03-23:
Good catch!   I see from looking at readSensor() that it sets internal CO2, temperature and humidity to 10000 if no data was read from the sensor.  I hadn't spotted that before.  Knowing that indicates an error condition I will add a test so post_dweet() doesn't get called if that error (no data) exists.  Thanks.

disquisitioner, 2022-03-23:
I added the required if() test in the network update branch I'm working on, so will follow up here when that pull request goes in.  (I added that same logic to apply to uploading data to InfluxDB.)

ericklein, 2022-05-12:
I modified the logic due to another bug for DWEET, MQTT, and Influxdb, and verified in MQTT and Influxdb that values publish when CO2 = 10000 (no CO2 sensor) and when we have an actual value. I'm going to close this bug. @disquisitioner you can reopen if there is an issue.

closed 2022-05-12 by ericklein